sony/flutter-elinux #251: How to configure DDS and DevTools for special target device

nguyenlkdn, 2024-04-04:
When attempting to connect to the VM service URI output by the Flutter engine in the context of a custom embedder, users may encounter the following error:

`This VM does not have a registered Dart Development Service (DDS) instance and is not currently serving Dart DevTools.`

**Platform:**
- IMX8QXP
- IMX-WAYLAND

**Flutter elinux release 3.13.0**

**Flutter target devices:**

> flutter-elinux devices
> 4 connected devices:
> 
> Linux (desktop)  • linux          • linux-x64      • Ubuntu 20.04.6 LTS 5.15.0-101-generic
> eLinux (desktop) • elinux-wayland • flutter-tester • Ubuntu 20.04.6 LTS 5.15.0-101-generic
> eLinux (desktop) • elinux-x11     • flutter-tester • Ubuntu 20.04.6 LTS 5.15.0-101-generic
**> eLinux (mobile)  • gridania-pvt1  • flutter-tester • ion-imx-wayland**

**Step to Reproduce:**
1) flutter-elinux create sample
2) cd sample
3) flutter-elinux pub get
4) flutter-elinux run -d gridania-pvt1 --profile -v

> [  +32 ms] Unable to locate an Android SDK.
> [  +10 ms] executing: uname -m
> [   +2 ms] Exit code 0 from: uname -m
> [        ] x86_64
> [  +89 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
> [        ] Artifact Instance of 'LegacyCanvasKitRemover' is not required, skipping update.
> [        ] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
> [   +5 ms] Artifact Instance of 'ELinuxEngineArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'MaterialFonts' is not required, skipping update.
> [        ] Artifact Instance of 'GradleWrapper' is not required, skipping update.
> [        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
> [        ] Artifact Instance of 'LegacyCanvasKitRemover' is not required, skipping update.
> [        ] Artifact Instance of 'FlutterSdk' is not required, skipping update.
> [        ] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
> [        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'FontSubsetArtifacts' is not required, skipping update.
> [        ] Artifact Instance of 'PubDependencies' is not required, skipping update.
> [   +5 ms] executing: ping -w 500 -c 1 192.168.7.1
> [  +11 ms] PING 192.168.7.1 (192.168.7.1) 56(84) bytes of data.
>            64 bytes from 192.168.7.1: icmp_seq=1 ttl=64 time=0.309 ms
>            1 packets transmitted, 1 received, 0% packet loss, time 0ms
>            rtt min/avg/max/mdev = 0.309/0.309/0.309/0.000 ms
> [   +3 ms] Skipping pub get: version match.
> [  +36 ms] Generating
> /home/saydi/workdir/00_Sources/community/flutter/sample/android/app/src/main/java/io/flutter/plugins/GeneratedPluginRegistrant.java
> [  +19 ms] Launching lib/main.dart on eLinux in profile mode...
> [   +1 ms] Uninstall sample from gridania-pvt1.
> [        ] executing: ssh root@192.168.7.1 rm -rf "/tmp/sample"
> [ +229 ms] Uninstallation Success
> [   +2 ms] Install sample (build/elinux/arm64/profile/bundle) to gridania-pvt1
> [        ] executing: scp -r build/elinux/arm64/profile/bundle root@192.168.7.1:/tmp/sample
> [+2970 ms] Installation Success
> [   +1 ms] Launch sample.name on gridania-pvt1
> [ +317 ms] VM Service URL on device: http://127.0.0.1:43621/d4EXYYZKy2c=/
> [        ] executing: ssh -o ExitOnForwardFailure=yes -L 127.0.0.1:43621:127.0.0.1:43621 root@192.168.7.1

**Result on Web-browser: debug and profile modes are the same result**

[image]

nguyenlkdn, 2024-04-09:
Already tried start VM Dart on Host PC to connecting VM Server URL
[image]

HidenoriMatsubayashi, 2024-04-09:
Please ensure that you have set forwardPort and disable all proxy settings.

See https://github.com/sony/flutter-elinux/wiki/Remote-target-devices#flutter_custom_devicesjson-example

nguyenlkdn, 2024-04-10:
> Please ensure that you have set forwardPort and disable all proxy settings.
> 
> See https://github.com/sony/flutter-elinux/wiki/Remote-target-devices#flutter_custom_devicesjson-example

```
{
  "custom-devices": [
    {
      "id": "gridania-pvt1",
      "label": "Gridania Pvt 1",
      "sdkNameAndVersion": "ion-imx-wayland",
      "enabled": true,
      "platform": "arm64",
      "backend": "wayland",
      "ping": [
        "ping", "-w", "500", "-c", "1", "192.168.7.1"
      ],
      "pingSuccessRegex": "ttl=",
      "install": [
        "scp", "-r", "${localPath}", "root@192.168.7.1:/tmp/${appName}"
      ],
      "uninstall": [
        "ssh", "root@192.168.7.1", "rm -rf \"/tmp/${appName}\""
      ],
      "runDebug": [
        "ssh", "root@192.168.7.1", "XDG_RUNTIME_DIR=/run/user/0 /tmp/${appName}/${appName} -b ."
      ],
      "stopApp": [
        "ssh", "root@192.168.7.1",
        "ps aux | grep \"/tmp/${appName}\" | grep -v grep | awk '{print $2}' | xargs kill"
      ],
      "forwardPort": [
        "ssh", "-o", "ExitOnForwardFailure=yes",
        "-L", "127.0.0.1:${hostPort}:127.0.0.1:${devicePort}", "root@192.168.7.1"
      ],
      "forwardPortSuccessRegex": "Linux"
    }
  ]
}
```

nguyenlkdn, 2024-04-10:
```
$ flutter-elinux run -d raspberry-pi4
Launching lib/main.dart on eLinux in debug mode...
Uninstall sample from raspberry-pi4.
hidenori@192.168.0.10's password: 
Uninstallation Success
Install sample (build/elinux/arm64/debug/bundle) to raspberry-pi4
hidenori@192.168.0.10's password: 
Installation Success
Launch sample.name on raspberry-pi4
hidenori@192.168.0.10's password: 
hidenori@192.168.0.10's password: 
Syncing files to device eLinux...                                  202ms

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

💪 Running with sound null safety 💪

An Observatory debugger and profiler on eLinux is available at: http://127.0.0.1:43237/xd6sW5hdEoE=/
Activating Dart DevTools...                                      1,830ms
The Flutter DevTools debugger and profiler on eLinux is available at:
http://127.0.0.1:9100?uri=http://127.0.0.1:43237/xd6sW5hdEoE=/
```

Regarding to your trial on Rpi, i just confused about why did your target board already installed "Dart DevTools" and it will be activated automatically while executing flutter app.

nguyenlkdn, 2024-04-10:
Practice done already, with below notes: 

I need to attach "--debug-url" manually after application started

Target Device:

```
LD_LIBRARY_PATH=/usr/share/flutter/3.19.3/profile/lib/ flutter-client --bundle=/usr/share/flutter/ion-hmi/3.19.3/profile/
flutter: The Dart VM service is listening on http://127.0.0.1:38867/LdVG8OhkIx8=/
```
Host PC: 
**Need to go into the project folder**
**Need to be sure PortForward already started before starting Host Dart Devtool**
Ex: in my case is `ssh -o ExitOnForwardFailure=yes -L 127.0.0.1:38867:127.0.0.1:38867 root@192.168.7.1 `
```
flutter-elinux attach --debug-url=http://127.0.0.1:38867/LdVG8OhkIx8=/
Syncing files to device eLinux...                                   6.2s

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

A Dart VM Service on eLinux is available at: http://127.0.0.1:43373/-fXmCK4fbG0=/
The Flutter DevTools debugger and profiler on eLinux is available at: http://127.0.0.1:9100?uri=http://127.0.0.1:43373/-fXmCK4fbG0=/
```

Results on Web Browser
[image]

Practice on Flutter-eLinux release 3.9.13. Thank you!

@HidenoriMatsubayashi I am not sure why did my environment haven't started Devtools automatically as your practiced

HidenoriMatsubayashi, 2024-04-10:
I've confirmed that `flutter-elinux` with a remote-device works fine.

flutter-elinux version: 3.19.4
Remote device: Rapsberry-Pi5

```
 flutter-elinux run -d raspberry-pi5 --profile
Launching lib/main.dart on eLinux in profile mode...
Uninstall sample from raspberry-pi5.
hidenori@192.168.10.24's password: 
Uninstallation Success
Install sample (build/elinux/arm64/profile/bundle) to raspberry-pi5
hidenori@192.168.10.24's password: 
Installation Success
Launch sample.name on raspberry-pi5
hidenori@192.168.10.24's password: 
hidenori@192.168.10.24's password: 

Flutter run key commands.
h List all available interactive commands.
c Clear the screen
q Quit (terminate the application on the device).
A Dart VM Service on eLinux is available at: http://127.0.0.1:39075/I6EFzjm6EeE=/
The Flutter DevTools debugger and profiler on eLinux is available at:
http://127.0.0.1:9100?uri=http://127.0.0.1:39075/I6EFzjm6EeE=/
```

[image]

nguyenlkdn, 2024-04-11:
Ya i mean is we able to connect for debugging on target already after we attach debugger manuallly

`flutter-elinux attach --debug-url=<Link to VM Service on target board>`

I just concerned is why did it is not started automatically on our environment. 
Maybe it is environment issues, thank for your confirmed.

HidenoriMatsubayashi, 2024-04-11:
Please try the followings to break down the issue.

1. Use the latest version of flutter-elinux
2. Use other devices such as Raspberry-Pi4/5

HidenoriMatsubayashi, 2024-04-15:
I'll close this issue as there is no additional update/comment.

HidenoriMatsubayashi, 2024-04-15:
Re-opening.

WDoughty, 2024-04-15:
I used that url from the verbose log and did the same process with running the `flutter-elinux attach` from my development machine and I was able to get dev tools. Not sure why its not running the attachment by itself. Here is the forwardPort for this custom device.

```
"forwardPort": [
        "ssh", "-o", "ExitOnForwardFailure=yes",
        "-L", "127.0.0.1:${hostPort}:127.0.0.1:{$devicePort}", "ccs@<display IP>"
      ],
"forwardPortSuccessRegex": "Linux"
```

HidenoriMatsubayashi, 2024-04-15:
> "-L", "127.0.0.1:${hostPort}:127.0.0.1:{$devicePort}", "ccs@<display IP>"

Typo: {$devicePort}
=> Correct one is ${devicePort}

WDoughty, 2024-04-16:
Good catch, but that is also how i have it in my custom devices. Working between different devices