sony/flutter-elinux #252: Error waiting for a debug connection: The log reader stopped unexpectedly.

xiaoxiaowesley, 2024-04-09:
When I run in my orangepi it show error below:

```
orangepi@orangepi3b:code/sample $ flutter-elinux run -d elinux-wayland
Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure you trust this source!
Launching lib/main.dart on eLinux in debug mode...
Building an eLinux application with wayland backend in debug mode for arm64 target...        74.4s
[ERROR][ELinuxWindowWayland(1066)] Failed to connect to the Wayland display.
[ERROR][CreateRenderSurface(1338)] Wayland display is invalid.
Error waiting for a debug connection: The log reader stopped unexpectedly.
Error launching application on eLinux.
```

and ` flutter-elinux devices ` exist devices:

```
orangepi@orangepi3b:code/sample $ flutter-elinux devices
Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure
you trust this source!
Found 4 connected devices:
  Linux (desktop)  • linux          • linux-arm64    • Ubuntu 22.04.4 LTS
  5.10.160-rockchip-rk356x
  Chrome (web)     • chrome         • web-javascript • Chromium 110.0.5481.4
  Ubuntu 22.04
  eLinux (desktop) • elinux-wayland • flutter-tester • Ubuntu 22.04.4 LTS
  5.10.160-rockchip-rk356x
  eLinux (desktop) • elinux-x11     • flutter-tester • Ubuntu 22.04.4 LTS
  5.10.160-rockchip-rk356x

```

Any one can help?

nguyenlkdn, 2024-04-09:
Can you try to check that you can execute "weston-simple-egl" on your board console.

xiaoxiaowesley, 2024-04-09:
> Can you try to check that you can execute "weston-simple-egl" on your board console.

Nope.  Command not found 

```
orangepi@orangepi3b:sample/build $ weston-simple-egl
zsh: command not found: weston-simple-egl
```

HidenoriMatsubayashi, 2024-04-09:
```
[ERROR][ELinuxWindowWayland(1066)] Failed to connect to the Wayland display.
[ERROR][CreateRenderSurface(1338)] Wayland display is invalid.
```

It seems it's wayland settings issue on your target board. Please check it.

nguyenlkdn, 2024-04-10:
> orangepi@orangepi3b:code/sample $ flutter-elinux devices
> Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure
> you trust this source!
> Found 4 connected devices:
>   Linux [...]

It looks like your rootfs is ubuntu and its' default display server is x11, not Wayland without additional installing.

xiaoxiaowesley, 2024-04-10:
> ```
> [ERROR][ELinuxWindowWayland(1066)] Failed to connect to the Wayland display.
> [ERROR][CreateRenderSurface(1338)] Wayland display is invalid.
> ```
> 
> It seems it's wayland settings issue on your target board. Please check it.

What settings need to be modified?

nguyenlkdn, 2024-04-10:
> What settings need to be modified?

```
sudo apt-get update
sudo apt-get -y install weston
weston &

# Test weston already started successfully
weston-simple-egl
```

xiaoxiaowesley, 2024-04-10:
> ```
> sudo apt-get update
> sudo apt-get -y install weston
> weston &
>
> # Test weston already started successfully
> weston-simple-egl
> ```

Thank you for reply.

 I Installed weston-simple-egl. but  when I  run `weston-simple-egl `, it show "The weston-simple-egl has stoped unexpectly".

I also try to rerun 'flutter-elinux run -d elinux-wayland'. 
Same thing happened

```
orangepi@orangepi3b:code/sample $ flutter-elinux run -d elinux-wayland
Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure
you trust this source!
Launching lib/main.dart on eLinux in debug mode...
Building an eLinux application with wayland backend in debug mode for arm64 target...       102.9s
[ERROR][ELinuxWindowWayland(1066)] Failed to connect to the Wayland display.
[ERROR][CreateRenderSurface(1338)] Wayland display is invalid.
Error waiting for a debug connection: The log reader stopped unexpectedly.
Error launching application on eLinux.

```

nguyenlkdn, 2024-04-11:
You have to be sure that `weston-simple-egl` app able to work well before elinux-wayland can work. 

`weston & `=> Did you see the weston window open after this command?

It looks like this one
[image]

HidenoriMatsubayashi, 2024-04-15:
I'll close this issue as there is no additional update/comment.

closed 2024-04-15 by HidenoriMatsubayashi