sony/flutter-elinux #255: Error when run flutter-elinux build elinux

xiaoxiaowesley, 2024-04-30:
When I run  `flutter-elinux build elinux ` in flutter project . It show error below:

```
The Dart DevTools debugger and profiler is available at: http://127.0.0.1:58095/aqAjkCPLitc=/devtools?uri=ws://127.0.0.1:58095/aqAjkCPLitc=/ws
Flutter assets will be downloaded from https://storage.flutter-io.cn/https://github.com/. Make sure you trust this source!
Downloading elinux-common tools...                                  12ms

'package:flutter_tools/src/cache.dart': Failed assertion: line 1193 pos 7: 'isAllowedUrl': URL not allowed: https://github.com/sony/flutter-embedded-linux/releases/download/c4cd48e186/elinux-common.zip
Allowed URLs must be based on one of: https://storage.flutter-io.cn/https://github.com/, https://storage.flutter-io.cn, https://storage.flutter-io.cn/flutter_infra_release/cipd
#0      _AssertionError._doThrowNew (dart:core-patch/errors_patch.dart:51:61)
#1      _AssertionError._throwNew (dart:core-patch/errors_patch.dart:40:5)
#2      ArtifactUpdater._download (package:flutter_tools/src/cache.dart:1193:7)
#3      ArtifactUpdater._downloadArchive (package:flutter_tools/src/cache.dart:1104:15)
#4      ArtifactUpdater.downloadZipArchive (package:flutter_tools/src/cache.dart:1065:12)
#5      ELinuxEngineArtifacts.updateInner (package:flutter_elinux/elinux_cache.dart:154:29)
#6      CachedArtifact.update (package:flutter_tools/src/cache.dart:848:11)
#7      Cache.updateAll (package:flutter_tools/src/cache.dart:715:24)
```

Solution is Simple.

```
export FLUTTER_STORAGE_BASE_URL=https://github.com/
```

In my envrionment variable such in ~/.zshrc (When using zsh)

It might be necessary to document this configuration in the README.md so that everyone is aware of it.

HidenoriMatsubayashi, 2024-05-01:
This is a known issue for developers in China. Updated https://github.com/sony/flutter-elinux/wiki/Use-local-elinux-artifacts-without-downloading-from-GitHub a little.

closed 2024-05-01 by HidenoriMatsubayashi

xiaoxiaowesley, 2024-05-06:
Perhaps there has been a misunderstanding. I can access to GitHub in terminal , and I have already set the FLUTTER_STORAGE_BASE_URL to https://storage.flutter-io.cn which is compatible with the network environment in China.

 The error is occurring elsewhere.