sony/flutter-elinux #270: Documentation Request: Guidance on Flutter-elinux and Third-Party Plugin Compatibility

Losses, 2024-09-13:
Firstly, thank you for maintaining such an amazing project. flutter-elinux has greatly facilitated developers in using Flutter within a Linux environment. However, I have encountered some issues while trying to integrate flutter-elinux into my Linux Flutter application, specifically regarding compatibility with third-party plugins.

## Issue Description

I am using [rinf](https://github.com/cunarist/rinf), a project that integrates Flutter and Rust. However, I am facing issues with loading dynamic libraries when trying to run the application. The specific error is as follows:

```
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Invalid argument(s): Failed to load dynamic library 'libhub.so': libhub.so: cannot open shared object file: No such file or directory
#0      _open (dart:ffi-patch/ffi_dynamic_library_patch.dart:11:43)
#1      new DynamicLibrary.open (dart:ffi-patch/ffi_dynamic_library_patch.dart:22:12)
#2      loadRustLibrary (package:rinf/src/load_os.dart:20:26)
#3      rustLibrary (package:rinf/src/load_os.dart:48:21)
#4      rustLibrary (package:rinf/src/load_os.dart)
#5      prepareInterfaceReal (package:rinf/src/interface_os.dart:20:3)
#6      initializeRust (package:rinf/rinf.dart:17:9)
#7      main (package:player/main.dart:31:9)
<asynchronous suspension>
```

Despite [trying multiple solutions](https://github.com/Losses/rinf/), I have not been able to resolve this issue. This process has been quite frustrating.

#### Request Details

To assist developers in better utilizing flutter-elinux and integrating third-party plugins (such as rinf) within a Linux environment, I would like to request some documentation guidance. Specifically, I am looking for:

1. Modifying Existing Packages: Instructions on how to modify existing packages to support the flutter-elinux platform.
2. **Example Projects**: If possible, an example project that includes third-party plugins (such as rinf) and demonstrates how to run them correctly in the flutter-elinux environment.

Thank you very much for your assistance and support. Providing the above documentation would greatly benefit many developers facing similar challenges.

Once again, thank you for your hard work!

Losses, 2024-09-14:
Inspected some detail about the cmake file:

https://github.com/sony/flutter-elinux/blob/70c11a81c630c1eb0bb528dc6d7b9e6dbbdbc08c/lib/elinux_plugins.dart#L571-L594
https://github.com/flutter/flutter/blob/1dce0c7ffb9e9686876400d69a1d26985a43169e/packages/flutter_tools/lib/src/flutter_plugins.dart#L572-L602

The FFI section is not generated

Losses, 2024-09-14:
OK, everything is clear now, I tried to fix that by comparing with the official flutter implementation:

https://github.com/Losses/flutter-elinux/blob/main/lib/elinux_plugins.dart

Maybe we should update the internal flutter implementation in this repo, since some code is pretty old and not compatible with "modern flutter"

Cheers

Losses, 2024-09-14:
[image]

I have confirmed that it will work after implementing the modifications I mentioned before on x11. However, making changes to `generated_plugin_registrant` is risky and could potentially disrupt the build system. To be honest, I have limited knowledge of C or C++ and have done my best with this task.

temeddix, 2024-09-14:
Thank you very much for the investigation and PR! I will do a review on it as soon as I have time.

Since it's a public holiday in South Korea, I might be busy with other schedules. Thank you again for your contribution and patience.

HidenoriMatsubayashi, 2024-09-14:
> Maybe we should update the internal flutter implementation in this repo, since some code is pretty old and not compatible with "modern flutter"

@Losses Thank you for reporting this issue. Would it be possible for you to create a PR? I'll review it.

closed 2024-09-15 by HidenoriMatsubayashi

yanshouwang, 2024-09-30:
Is this fixed already? I still see this error when write plugin with dart:ffi.
``` bash
orangepi@orangepi5max:~/dev/hybrid_sdk/hybrid_v4l2/example$ flutter-elinux run -d elinux-x11
Launching lib/main.dart on eLinux in debug mode...
Building an eLinux application with x11 backend in debug mode for arm64 target...        16.7s
arm_release_ver: g13p0-01eac0, rk_so_ver: 10
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Invalid argument(s): Failed to load dynamic library 'libhybrid_v4l2.so': libhybrid_v4l2.so: cannot open shared object file: No such file or directory
#0      _open (dart:ffi-patch/ffi_dynamic_library_patch.dart:11:43)
#1      new DynamicLibrary.open (dart:ffi-patch/ffi_dynamic_library_patch.dart:22:12)
#2      _dylibHybridV4L2 (package:hybrid_v4l2/src/ffi.x.dart:18:45)
#3      _dylibHybridV4L2 (package:hybrid_v4l2/src/ffi.x.dart)
#4      libHybridV4L2 (package:hybrid_v4l2/src/ffi.x.dart:20:41)
#5      libHybridV4L2 (package:hybrid_v4l2/src/ffi.x.dart)
#6      V4L2Impl.open (package:hybrid_v4l2/src/v4l2_impl.dart:45:20)
#7      main (package:hybrid_v4l2_example/main.dart:9:19)
#8      _runMain.<anonymous closure> (dart:ui/hooks.dart:301:23)
#9      _delayEntrypointInvocation.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:297:19)
#10     _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:184:12)
```

I can find the .so in build dir.

Losses, 2024-09-30:
@yanshouwang Probably you need to update your package by adding a elinux platform support, check this:

https://github.com/cunarist/rinf/pull/435/files#diff-faf9417d945a8218c774581836aba31ca11b1a1a1818374d630c5fc1bb21a148R48-R49

yanshouwang, 2024-09-30:
I've update the CMakeLists.

elinux/CMakeLists.txt
``` camke
cmake_minimum_required(VERSION 3.10)
set(PROJECT_NAME "hybrid_v4l2")
project(${PROJECT_NAME} LANGUAGES CXX)

# Invoke the build for native code shared with the other target platforms.
# This can be changed to accommodate different builds.
add_subdirectory("${CMAKE_CURRENT_SOURCE_DIR}/../src" "${CMAKE_CURRENT_BINARY_DIR}/shared")

# This value is used when generating builds using this plugin, so it must
# not be changed
set(PLUGIN_NAME "hybrid_v4l2_plugin")

add_library(${PLUGIN_NAME} SHARED
  "hybrid_v4l2_plugin.cc"
)
apply_standard_settings(${PLUGIN_NAME})
set_target_properties(${PLUGIN_NAME} PROPERTIES
  CXX_VISIBILITY_PRESET hidden)
target_compile_definitions(${PLUGIN_NAME} PRIVATE FLUTTER_PLUGIN_IMPL)
target_include_directories(${PLUGIN_NAME} INTERFACE
  "${CMAKE_CURRENT_SOURCE_DIR}/include")
target_link_libraries(${PLUGIN_NAME} PRIVATE flutter flutter_wrapper_plugin)

# List of absolute paths to libraries that should be bundled with the plugin
set(hybrid_v4l2_bundled_libraries
  # Defined in ../src/CMakeLists.txt.
  # This can be changed to accommodate different builds.
  $<TARGET_FILE:hybrid_v4l2>
  PARENT_SCOPE
)
```

src/CMakeLists.txt
``` cmake
# The Flutter tooling requires that developers have CMake 3.10 or later
# installed. You should not increase this version, as doing so will cause
# the plugin to fail to compile for some customers of the plugin.
cmake_minimum_required(VERSION 3.10)

project(hybrid_v4l2_library VERSION 0.0.1 LANGUAGES C)

add_library(hybrid_v4l2 SHARED
  "hybrid_v4l2.c"
)

set_target_properties(hybrid_v4l2 PROPERTIES
  PUBLIC_HEADER hybrid_v4l2.h
  OUTPUT_NAME "hybrid_v4l2"
)

target_compile_definitions(hybrid_v4l2 PUBLIC DART_SHARED_LIB)
```

I can fix this by change the .so lib path to absolute.

``` Dart

// final _dylibHybridV4L2 = ffi.DynamicLibrary.open('libhybrid_v4l2.so');
final _dylibHybridV4L2 = ffi.DynamicLibrary.open(
    '/home/orangepi/dev/hybrid_sdk/hybrid_v4l2/example/build/elinux/arm64/debug/bundle/lib/libhybrid_v4l2.so');

```

PR #271 (2024-09-15): fix: Unable to Build Projects with FFI Plugins in flutter-elinux
#### Background
Flutter has modified the way it retrieves plugins. The `messageChannel` plugins and `ffi` plugins are now retrieved and imported separately. Currently, the internal implementation of `flutter-elinux` has not been updated to reflect this change, resulting in build failures for plugins that depend on FFI, such as `cargokit` and `rinf`.

#### Solution
This PR references the official Flutter implementation to provide a fix for this issue. The changes ensure that plugins relying on FFI are correctly compiled and proces [... cut]