sony/flutter-elinux #281: Flutter 3.29.0

RagavL, 2025-02-24:
When will flutter 3.29.0 be supported?

HidenoriMatsubayashi, 2025-02-24:
@taka-okayama

taka-okayama, 2025-02-28:
@RagavL @HidenoriMatsubayashi 

We've been encountering a build issue in CI/CD in the following flutter stable version. Under research now.

Version: https://raw.githubusercontent.com/flutter/flutter/stable/bin/internal/engine.version

mmerah, 2025-05-06:
Is this still being looked at? I know this project was forked from flutter-tizen, which now has flutter 3.29.0 support. Would that help?

taka-okayama, 2025-05-09:
@mmerah 
thank you for your helpful information! I'll check it!

I found this PR.
https://github.com/flutter-tizen/flutter-tizen/commit/728ad1cc887bf044ee90ec86db8d7232efc89371

lhoward, 2025-06-08:
Also wondering if there'll be a tag for 3.32.2?

RagavL, 2025-09-22:
Any update on release of 3.29.3 ?