sony/flutter-elinux #289: About the next version update schedule

makotosato-at, 2025-07-03:
Hi there,

Thanks for all your hard work on this project.
I've been using flutter-elinux and it's been really helpful for my development work.
I noticed that releases seem to have slowed down recently, and I was wondering when the next version update might be coming?
If possible, could you share any upcoming release plans or roadmap? It would really help with my project planning.
Sorry to bother you when you're probably busy, but I'd appreciate any info you can share.

Thanks.

HidenoriMatsubayashi, 2026-01-13:
Adding @taka-okayama 

I would have liked to continue maintaining this project if I was still a maintainer, but Sony decided to remove it from me. Therefore, I don't intend to work on it at this time. Sorry about that.

martinetd, 2026-01-21:
@taka-okayama 
> Could you give me a little time? Some parts of our internal CI contain code that we cannot share externally, so I need to review what we can provide.

Sure, thank you!
I can copy-paste the initial release to get started with a commit renaming the repos, but we'll need releases to get started with any fix, so anything you can share will be appreciated.

There haven't been any release in a while though so this can definitely wait a few days, please do as you can.

@lhoward 
> I have a few patches sitting [flutter-embedded-linux](https://github.com/PADL/flutter-embedded-linux) but I think I have submitted most of them as PRs, and many have been merged. HiDPI fixes is probably the outstanding one.

Yes, thank you! I saw the flutter-embedded-linux PRs, it looks good to me but I'd like to test, and I'm afraid I'm not quite sure how to test right now short of building both a flutter app and flutter-embedded-linux separately and copying the .so by hand in the build project 😅
I don't want to be too eager and merge too much too fast, so will take a bit more time to get started than I intended to but I'll get to it (the comment about elinux-common.zip in https://github.com/sony/flutter-embedded-linux/pull/437#issuecomment-3106336952 also made me curious and I'd like to understand that better too -- there probably is a way to manage the flutter-elinux cache to bypass zips and use a local repo I haven't found...)

> https://github.com/flutter-elinux sounds good, definitely happy to help work on this, is anyone interested in a Zoom call sometime next week?

Oh, I was about to refuse based on timezones but you're actually on a Japan-friendly TZ 😁
My personal preference is to get thing through writing because I've always have a hard time making calls work as soon as USA/Europe/(Japan|Australia) are involved, and most of my coworkers don't handle spoken English so well, but if you think it'd be helpful to get started I'll be happy to join.
Would you happen to use matrix or IRC to have a chat for the rest of the time?

I made you co-owner of the org for now, there really isn't anything more than the forked repos so feel free to muck about if you'd like but I don't think we need to rush this; let's take some time to get organized first.

lhoward, 2026-01-21:
Agree 💯. Let's take our time and make sure all changes go through PRs and are reviewed by at least us both. No need to jump on a call, but also happy just to have a chat on your medium of choice next week to introduce ourselves. Or not 😃

martinetd, 2026-01-22:
Great, I'll read the scripts and give it a shot by the end of the week! Thank you, this is appreciated.